LucidVR/lucidgloves #1: Assembly Guide Prototype 3

Lobotuerk, 2021-04-22:
Do you have a video or a guide to assembly prototype 3? The videos on your channel show one of the older ones, and I printed the parts for the 3.1 and have no idea how to assembly it. Thanks in advance

lucas-vrtech, 2021-04-24:
I am going to be posting a video tutorial on youtube soon.

CameronKlinge, 2021-05-02:
i need help with wiring, so I see that on the wiring demonstration it shows how to wire to the AO A1 and so on but the other 2 pins on the petameter I don't get how to wire those. btw im wiring it without a breadboard.

lucas-vrtech, 2021-05-02:
@CameronKlinge The other two pins correspond to GND and 5V. It needs to be split from the Arduino so that each potentiometer can get GND and 5V. (Or 3V3 if using an ESP32)

This guide may help:
https://www.arduino.cc/en/tutorial/potentiometer

CameronKlinge, 2021-05-02:
@lucas-vrtech Ok that makes more sense thank you so much!

danwillm, 2021-06-04:
Going to close this now that the tutorial is out, but please do feel free to re-open if you need more clarification.

Lobotuerk, 2021-06-04:
Thank you @danwillm , Awesome work

closed 2021-06-04 by Lobotuerk