LucidVR/lucidgloves #4: Plans to support Wifi communication?

Nitwel, 2021-06-04:
Are there any plans on supporting this using websockets or something? Or are there reasons not to go this road like latency or complexity?

LittleNyanCat, 2021-11-14:
The main reason WI-FI hasn't been implemented yet is that using it on the ESP32 disables ADC2, leaving us with only 6 pins available for analog input, making tracking all fingers while having a joystick impossible without extra hardware.

You're free to implement it and contribute if you wish, but for now we don't have enough reasons to do so ourselves

minystreem, 2022-01-06:
I could try to program this, otherwise I can close this issue.

danwillm, 2022-01-06:
@minystreem 
You are very welcome to implement this feature, there's been a number of people who have suggested it.

minystreem, 2022-01-06:
Ok I will start on it, I may have to make changes to Lucid-ui

danwillm, 2022-01-06:
You shouldn't have to do much on the UI, it automatically loads all the settings in `default.vrsettings` file found in the driver. Though you can add your own custom titles and descriptions to the labels and tooltips.

The majority of the work will be on the driver and firmware. Let me know if you have problems with any of that.

minystreem, 2022-01-06:
Yea I meant the driver sry. I am having trouble finding a good library

danwillm, 2022-01-06:
I recommend ASIO
https://think-async.com/Asio/

minystreem, 2022-01-06:
does that work for the esp32

danwillm, 2022-01-06:
Sorry, linked the wrong library https://github.com/zaphoyd/websocketpp (this one is the one @Nitwel was originally using but I don't think he finished his implementation)

minystreem, 2022-01-06:
So that's the library I should use for the client side AND the sever side

danwillm, 2022-01-06:
Just driver, I'm sure there's other arduino libraries for wifi.

minystreem, 2022-01-06:
Ok, I will try to find one, the WiFi library is outdated so I might have to find a different one

LittleNyanCat, 2022-01-06:
For the Esp side I've used this one for websockets in one of my projects before:
https://github.com/me-no-dev/ESPAsyncWebServer

minystreem, 2022-01-06:
Ok thx

minystreem, 2022-01-13:
@LittleNyanCat would you like to help do the driver side of things? my c++ is not very good, I got the esp side kinda working, I'm still working on it I got a bit busy. If your could help that would be nice but if not that's fine.

LittleNyanCat, 2022-01-13:
Uhh, the Arduino side stuff is also C++ though?

I'd help right now if I didn't find setting stuff up to build the driver to be such a chore

I won't be touching any driver related stuff until my gyros arrive in about a month, when that happens maybe I'll take a look at it

minystreem, 2022-01-13:
Yea building stuff is a chore for too, I'll finish the esp side

Veillax135, 2022-01-13:
Sorry, not available right now. I would if I had the time.

-Veillax135

CaelanTheCoder, 2022-08-16:
There's like 66 forks of this project; pretty sure at least one person has implemented WIFI communication.

minystreem, 2022-12-11:
no, I kinda abanonded it becuse I dont get the point. most forks dont have any changes

minystreem, 2022-12-11:
Wifi communicaion is hard, you try doing it

AthemiS13, 2024-12-10:
> Wifi communicaion is hard, you try doing it

Hey, have you managed to get it working? I want to do something simmilar

ElectricTurtle32, 2024-12-10:
I got the aurdino side working, but not the Pc side, I would need to dig up
the code though lol

AthemiS13, 2024-12-10:
> I got the aurdino side working, but not the Pc side, I would need to dig up
> 
> the code though lol
> 
> 
> 
> 
> 
> 
> 
> > Wifi communicaion is hard, you try doing it
> 
> >
> 
> > Hey, have you managed to get it working? I want to do something simmilar

Could you send me your discord or something please so i can contact you privately?

ElectricTurtle32, 2024-12-16:
> #4 (comment)

I don't think I have the code anymore unfortunately, I am very sorry. I
wish you luck on your project though!

JohnRThomas, 2024-12-23:
@AthemiS13 The driver needs to be modified, but this is the start of serial over wifi that you can rip from my fork if you want to try it: https://github.com/JohnRThomas/OpenGloves-Firmware/blob/main/open-gloves/SerialWIFICommunication.hpp