LucidVR/lucidgloves #20: Inseam on spool too small

Rat-Boi-Bop, 2021-09-03:
I am a dumb person and would like help. The inseam on the spool stl file isn't showing up on the layer view, apparently it is so small my printer can't see it and so it just skips it. You can see how this is an issue. Please advise.

Zach710032, 2021-09-03:
same probelm as above^ adding z layer fixed it.

lucas-vrtech, 2021-09-04:
I will be throwing up some new STL files with release proto 4.0.1 which will fix this issue. Thank you for pointing it out. This is actually a problem carried from the original prototype 3 that was fixed in 3.1. For some reason the fix did not carry over to prototype 4 so I will do that manually.

wpuchalski, 2021-09-07:
I have the exact same issue I thought it was something I was doing with my printer. Glad to hear I'm not the only one. Thanks for all your work Lucas.

wpuchalski, 2021-09-10:
Just as a heads up I modeled up a new spool and am printing it now, if it works I could share the file with you Rat-Boi-Bop and Zach

lucas-vrtech, 2021-11-21:
Solved in prototype 4.1

closed 2021-11-21 by lucas-vrtech