LucidVR/lucidgloves #21: ValueError: could not convert string to float: '511&511&511&511&511&232&164&0&0&0&0&0&0'

ruchitsheth, 2021-09-12:
I am using the Prototype 1 Code for blender and when I am trying to run the code with a potentiometer as input, I am getting the following error

Read blend: C:\Users\Ruchit\Downloads\lucidgloves-blender-main\lucidgloves-blender-main\VRGloveDemo_Attempt2.blend
b'511&511&511&511&511&232&164&0&0&0&0&0&0\n'
Traceback (most recent call last):
  File "C:\Users\Ruchit\Downloads\lucidgloves-blender-main\lucidgloves-blender-main\VRGloveDemo_Attempt2.blend\script.py", line 43, in <module>
ValueError: could not convert string to float: '511&511&511&511&511&232&164&0&0&0&0&0&0'
Error: Python script failed, check the message in the system console

Please help ME!

lucas-vrtech, 2021-09-25:
This isn't a problem with lucidgloves firmware, rather it's something that needs to be fixed in the blender script. I'll leave this issue open to make sure I/someone fix it at some point.

fionn77, 2021-12-29:
in blender change this line :
lineTable = read.decode().split(',')           to
lineTable = read.decode().split('&')

fionn77, 2021-12-29:
use new lucidgloves-firmware and use this code in main instead of original:

#define ALWAYS_CALIBRATING CALIBRATION_LOOPS == -1

String CommandFromPC;
static char stringToPrint[75];

ICommunication* comm;
int loops = 0;

void setup() {
  // Command string for serial interface
  CommandFromPC = "";

  
  #if COMMUNICATION == COMM_SERIAL
    comm = new SerialCommunication();
  #elif COMMUNICATION == COMM_BTSERIAL
    comm = new BTSerialCommunication();
  #endif  
  comm->start();

  setupInputs();
  
}

void loop() {
//  if (comm->isOpen()){
    #if USING_CALIB_PIN
    bool calibButton = getButton(PIN_CALIB);
    if (calibButton)
      loops = 0;
    #else
    bool calibButton = false;
    #endif
    
    bool calibrate = false;
    if (loops < CALIBRATION_LOOPS || ALWAYS_CALIBRATING){
      calibrate = true;
      loops++;
    }
    
    int* fingerPos = getFingerPositions(calibrate, calibButton);
//    int* fingerPos;

    bool joyButton = getButton(PIN_JOY_BTN);

    #if TRIGGER_GESTURE
    bool triggerButton = triggerGesture(fingerPos);
    #else
    bool triggerButton = getButton(PIN_TRIG_BTN);
    #endif

    bool aButton = getButton(PIN_A_BTN);
    bool bButton = getButton(PIN_B_BTN);

    #if GRAB_GESTURE
    bool grabButton = grabGesture(fingerPos);
    #else
    bool grabButton = getButton(PIN_GRAB_BTN);
    #endif

    #if PINCH_GESTURE
    bool pinchButton = grabGesture(fingerPos);
    #else
    bool pinchButton = getButton(PIN_PNCH_BTN);
    #endif

    //  OUTPUT  
    //    comm->output(encode(fingerPos, getJoyX(), getJoyY(), joyButton, triggerButton, aButton, bButton, grabButton, pinchButton));

    CommandFromPC = Serial.readStringUntil('\n');
    if(CommandFromPC == "Hello Arduino") {
      Serial.println("What is your Bidding my Master"); 
    } else if (CommandFromPC == "READ_VALUES") {
//       Serial.println( encode(fingerPos, getJoyX(), getJoyY(), joyButton, triggerButton, aButton, bButton, grabButton, pinchButton) );

//         fingerPos = getFingerPositions(calibrate, calibButton);

         sprintf(stringToPrint, "%d&%d&%d&%d&%d\n",  fingerPos[4], fingerPos[3], fingerPos[2], fingerPos[1], fingerPos[0]  );

//         Serial.println(stringToPrint);
         Serial.print(stringToPrint);
         Serial.flush();

         
    }

    
    delay(LOOP_TIME);
//  }
}

after that you can delete Encoding tab

CaelanTheCoder, 2022-08-16:
this issue is outdated right?

CaelanTheCoder, 2022-08-16:
this can safely be closed right?