LucidVR/lucidgloves #22: ESP32s not sending/recieving

JamesPCVR, 2021-09-12:
I have uploaded the software to both gloves and have connected them to my computer over Bluetooth. The opengloves driver is able to track their position but I have no button inputs, finger tracking or haptic feedback. I am able to perform auto-calibration in the driver but not extend/retract the servos fully, I'm just told the command could not be added to the pipeline.

The configuration in the opengloves driver is as follows:
Both gloves enabled
Communication method Bluetooth serial
Right name: lucidgloves-right (same for firmware and serial number)
Left name: lucidgloves-left (same for firmware and serial number)
Driver emulation: lucid glove
Encoding protocol: Alpha

Any fixes?

lucas-vrtech, 2021-09-25:
Are you able to check the bluetooth output in arduino IDE over bluetooth serial port numbers?

Zach710032, 2021-09-26:
Change your comms method to usb serial and use the serial monitor(in arduino ide go to tools>serial monitor and then change baud rate to 115200) to see if your pots are actually sending positional data to the Esp32s and then to the computer. then, make sure you have the comms method changed back to btserial when you have verified the positional tracking works. Make sure opengloves is enabled in steam, and try flipping your pot values if you think you may have wired them backwards.

danwillm, 2021-09-27:
In addition to this, SteamVR must be open for those functions to work. In a later release, this won't be needed.