LucidVR/lucidgloves #25: Latency issue and loss of connection when using bluetooth

jschung1111, 2021-09-27:
After building my pair of gloves, I tried testing each of the gloves that I've made. The right glove worked wonderfully. However, when I went to check my left glove, I noticed significant latency issues and after a few seconds it would disconnect and I would have to restart Steam VR to get it to connect again (only to have it repeat the issue that I just mentioned). Any ideas on what's causing this issue?

danwillm, 2021-09-27:
Interesting, does this happen with both gloves when they're connected?

There's a couple of bluetooth fixes that may or may not be related to the issue you are having on the beta branch on Steam, could you try opting into it and see if that helps things (you can opt into the beta by right clicking opengloves in the Steam library list->properties->betas and select the beta branch.

Let me know if that fixes things.

Thanks.

lucas-vrtech, 2021-11-21:
Closing this issue now, but feel free to reopen if needed or ask around in the discord. Biggest recommendation I can give is to try out ESP32-WROOM-32U's with an external antenna as it helps with signal strength significantly

closed 2021-11-21 by lucas-vrtech