LucidVR/lucidgloves #33: My Hands are not responding to the finger movements in SteamVR home

hlaila1, 2021-11-07:
Hi There,

So I followed all the steps and installed all the drivers required. I've downloaded the latest Driver 0.4 release and changed the values to 1023 because I am using Nano. Also in the Driver settings I made sure the COM ports are the same as my Arduino. 

The "Serial Monitor" showed big and small numbers meaning it is working as Lucas has showed in the video. After I linked my Oculus Rift S, and launched SteamVR. It showed the Valve Controllers following my Oculus Controllers. When I am at "Steam HOME" My hands appears but they are not responding to the finger movements and also they are showing only 3 fingers and a Thumb!!

I am not sure what if I am missing something here, Please help :(

danwillm, 2021-11-07:
The three fingered hand in SteamVR Home is an intenional hand model made by the SteamVR developers, and does not respond to finger movement. In the SteamVR Home menu, change your hand model to the five fingered hand (by pressing the b button on the controller. You might need to deactivate the openglove driver first to use your primary controllers, either in SteamVR settings or in the openglove configuration file)

Did you download the driver on Steam? If you change the hand model to a five fingered hand and it does not respond to finger movement, please delete whatever you installed previously. There is a configuration ui in the steam release that makes it easier to configure settings.

hlaila1, 2021-11-08:
Thank you @danwillm appreciate helping me with this awesome project.
so I figured our how to change the hand styles. Now I still have the same issue of the fingers are not responding to the movement and it is just a fronzen hand floating and following the movement of my oculus controller.
I am not sure what is the problem, all the ports and the configurations are correct. :( :(
I've attached few picture of the glove. Please see below

I meant to ask you, and please feel free to say no, but I would like to schedule a call with you to go over the glove build if you don't mind of course. Also I would like to discuss with you and Lucas how can help you guys and support this project, please let me know.

[image]
[image]
[image]
.

danwillm, 2021-11-08:
Please make sure that you have installed the driver on steam, instead of downloading it via GitHub - it'll make life a lot easier for you with a ui for managing configuration options, and automatically installing the driver. If you do this, please make sure to remove whatever files you added to SteamVR previously for opengloves. Clicking launch after donwloading it on Steam will present a configuration interface to you where you can alter these settings.

If it's still causing you problems, feel free to attach the vrserver log files here and I can take a further look. `C:\Program Files (x86)\Steam\logs\vrserver.txt`

Lucas and I are pretty tight for time at the moment, so it might be hard to find a time that works, but feel free to send a message on the discord server, or a dm to either of us there. `danwillm#8254`

danwillm, 2021-12-28:
As we haven't heard back, I'm going to close this due to inactivity and I'm assuming this issue has been fixed.

If it hasn't, feel free to attach your log files either here or on discord, and we can reopen the issue.

closed 2021-12-28 by danwillm