LucidVR/lucidgloves #34: Spool cover and tensioner not connecting

AbsoluteXero132, 2021-11-08:
I printed the tensioner and spool cover pieces and they are not connecting properly. In the video he was able to put them on without issue. I was able to get them together, but they broke in the process. They were printed with a resin printer. Is that the problem? I tried using a 3D printer but it was having issues.

jlschaper, 2021-11-20:
I haven't printed the parts in resin (but was thinking of giving that a go) but make sure you've sanded any elephant footing etc off.  Also consider volumetric shrinkage when curing - can be up to 5 or 6%!!  If I can find time I might print a v4 set in resin and see how it goes too.

lucas-vrtech, 2021-11-21:
Another think to keep in mind (and I'm not sure how this translates to SLA printers, but) you want to make sure that you keep print the cover with the side that has a lip AWAY from the bed, because that's the part that is the most sensitive to elephant's foot.

DirtyBong, 2022-01-12:
I'm having the same issue, on the previous prototype they snapped in for me but for this print I've only gotten 1 out of 10 to snap-in. It might just be of my own doing but I'm not sure what would be causing it. I even printed them with the lip away from the bed as Lucas suggested and still didn't seem to help much.

jlschaper, 2022-01-14:
I found that if I clipped the ring on the tensioner *before* curing there wasn't the issue of snapping.  It requires a little more fiddling around to get the spring hook in, but once you start winding there was no difference.
If that isn't feasible, maybe take a sharp knife/scalpel and cut a notch in the lip of the tensioner and in the lip of the ring and then "screw"/twist the cover on rather than snapping it on.

closed 2022-07-12 by lucas-vrtech