LucidVR/lucidgloves #35: Arduino IMU Tracking?

QueasyQolon, 2021-11-09:
Would it be feasible to add Arduino-based tracking with a gyroscope/accelerometer sensor? That would allow for tracking that doesn't require existing VR controllers / trackers, since Vive tracker pucks are kinda pricey.

StephenIndustries, 2021-11-09:
Unfortunately, IMU tracking is both beyond the scope of this project, and wouldn't be nearly reliable enough to be a viable tracking method. As it stands, IMUs have too much inherent drift to viable in and of themselves as a tracking method.

LittleNyanCat, 2021-11-14:
This has been asked many times in our discord server and short answer is no. Even SlimeVR will most likely to be too innacurate/drifty for something that requires precision like your hands

lucas-vrtech, 2021-11-21:
Closing for the same reason LittleNyanCat mentioned. IMU based tracking could be used for wrist tracking, which would allow you to place your controller/tracker further up the arm, but that's about it unfortunately, until there's a really good DIY solution to counter IMU drift. Hopefully the SlimeVR guys can get it more accurate someday that we can work together on some cool hand tracking solution.

closed 2021-11-21 by lucas-vrtech