LucidVR/lucidgloves #36: OpenVR Init Error

NoavanBentem, 2021-11-17:
Using a Esp32 devkit v1 and following all steps closely it seems something isn't going as planned. Very unsure where the problem might be as I'm not yet so advanced in electronics(but getting better). What catched my eye is my serial output looks way different from what yours showed in one of your videos("How I build CHEAP VR Gloves.")

Mine looks more like this:
A1662B3329C1229D4095E4095F384G0ILM
A112B3994C305D4095E1017F336G34ILM
A2816B4037C3596D4095E1115F469G56ILM
A852B3344C240D4095E4095F349G0ILM

Just to be sure I tried a nano as well with same serial output

LittleNyanCat, 2021-11-17:
That video is relatively old and using what we now call the "legacy" encoding. You are using the alphanumeric encoding which is the current encoding we use

danwillm, 2021-12-28:
I'm going to close this due to inactivity. Like @LittleNyanCat mentioned, you should be using alpha encoding.

If you're running into issues that relate to your title "OpenVR Init Error" (or are on the pc side of things), that is something different to this, and you should open an issue on the [opengloves](https://github.com/LucidVR/opengloves-driver) repo.

closed 2021-12-28 by danwillm