LucidVR/lucidgloves #44: Error compiling for board ESP32 Wrover Module.

slobershelf, 2021-12-21:
when I try to download the program after following the steps it says   input[size] = NULL;
                   ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
c:/users/       /appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/users/         /appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Wrover Module.

LittleNyanCat, 2021-12-21:
Oh great, it's that annoying bug that we have no idea why it happens and not our fault again.

Anyhoo, the fix is uninstalling the IDE, nuking the Arduino15 folder in Appdata/Local and reinstalling everything again

slobershelf, 2021-12-21:
Thank u lol it’s been happening all day--
Nicholas Leite

slobershelf, 2021-12-21:
question how do you uninstalling the IDE and all that

LittleNyanCat, 2021-12-21:
Google how to uninstall programs, it's not that hard.

To delete the Arduino15 folder navigate to C:\Users\<username>\AppData\Local

slobershelf, 2021-12-21:
thank u <3

closed 2021-12-21 by slobershelf