LucidVR/lucidgloves #45: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

slobershelf, 2021-12-21:
esptool.py v3.0-dev
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

LittleNyanCat, 2021-12-21:
Press and hold the BOOT/IOO (whichever your board has) as soon as you press the upload button. Release when it says the upload is done. That should do it

slobershelf, 2021-12-21:
it got closer but it said 
esptool.py v3.0-dev
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:cc:a8:3f:6c:38
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.4 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

StephenIndustries, 2021-12-21:
Can you try **holding**  the button down until it finishes uploading?

slobershelf, 2021-12-21:
I did

slobershelf, 2021-12-21:
it did the same thing

LittleNyanCat, 2021-12-21:
What board do you have set in the IDE?

slobershelf, 2021-12-21:
where do you check

LittleNyanCat, 2021-12-21:
It's under tools

slobershelf, 2021-12-21:
ESP32 Dev module

LittleNyanCat, 2021-12-21:
Try switching that to Node32s

slobershelf, 2021-12-22:
same error

Sketch uses 949498 bytes (72%) of program storage space. Maximum is 1310720 bytes.
Global variables use 31172 bytes (9%) of dynamic memory, leaving 296508 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:cc:a8:3f:6c:38
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5957.9 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

Benjamin5678, 2022-01-18:
Have you downloaded these drivers? Its 4th in list https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

danwillm, 2022-03-06:
I'm assuming this issue is fixed. I'll close this issue now, but feel free to reopen if you're still running into problems.

closed 2022-03-06 by danwillm