LucidVR/lucidgloves #46: potentionometer wont work?

Philliboyxd, 2021-12-28:
hi guys i finished my left glove by using all the thinks lucas has. i wired everything to the esp32 wroom-32 v1 mpdul and when i go to the serial port und move my thumb and pint finger the numbers change but when i move my pinky or middle finger the numbers are all the same :( Pls help me and i wired everthing to the pins and change it in the ide to the pins that i use

LittleNyanCat, 2021-12-28:
What pins did you use? Some pins don't have ADCs and thus can't be used for the pots

Philliboyxd, 2021-12-28:
i used g34,g35,g32,g25,g26

LittleNyanCat, 2021-12-28:
Alright, all of those have ADCs. Double check your wiring then, that's always the problem everyone tends to have. Just because it looks good doesn't mean it is good

Philliboyxd, 2021-12-28:
ok i checkt all of it and all are conecctect perfect to the potetinomiter as well to the dev board
can there be any other problem?

LittleNyanCat, 2021-12-28:
How did you "checkt" it?

Philliboyxd, 2021-12-28:
i pull the wire and look at this if its connected to the board and potentionometer. is it wrong?

LittleNyanCat, 2021-12-28:
Check if there is an actual proper connection with a multimeter

Philliboyxd, 2021-12-28:
multimeter?

LittleNyanCat, 2021-12-28:
Yes. Multimeters are an essential tool when doing anything electronics related

Philliboyxd, 2021-12-28:
ah that the wires are not broken ?

Philliboyxd, 2021-12-28:
what problem culd be there anyways `?

closed 2021-12-28 by Philliboyxd