LucidVR/lucidgloves #50: OpenVR Init Error! Err: 110, Please kill SteamVR processes (VR Server) in Task Manager and try again.

Benjamin5678, 2022-01-16:
When I launch the OpenGloves app in steam I get this error. I checked task manager and had no "VR Server" or SteamVr proccess and I even killed all my background oculus tasks to be sure. I'm not sure what other information I should provide.

Benjamin5678, 2022-01-16:
Error was due to SteamVR not being installed. Pretty bad error message if you ask me.

closed 2022-01-16 by Benjamin5678

lucas-vrtech, 2022-01-16:
Glad you were able to figure it out. Indeed, SteamVR is one of the requirements for OpenGloves. I'll talk to Dan and see if there's any way we can better document that in the UI.

For future reference, this issue would be more appropriately placed in https://github.com/LucidVR/opengloves-ui/issues.

Thanks