LucidVR/lucidgloves #66: Right hand of prototype 4 does not have a download button

TechnicallyOffbeat, 2022-04-08:
Prototype 4's [left hand](https://github.com/LucidVR/lucidgloves/blob/main/hardware/Prototype4/RigidMount_leftHand.STL) rigid mount has a download button and seems to be working as intended.  
The [right hand](https://github.com/LucidVR/lucidgloves/blob/main/hardware/Prototype4/RigidMount_rightHand.STL) rigid mount does not have a download button, and is significantly larger for a file that should be basically the same as the left (76KB vs 374KB).

Burritoclock, 2022-04-14:
Same issue here

danwillm, 2022-04-14:
it is recommended to download the entire repository, then use the files from the download. You can download the repo by pressing the green code button on the homepage, then pressing download zip.

closed 2022-04-14 by danwillm