LucidVR/lucidgloves #68: Error while compliling code Look at comment for log

AikoKiss, 2022-04-13:
file included from C:\Users\AikoKiss\Downloads\Compressed\FlexiaGloves-main_2\FlexiaGloves-main\firmware\lucidgloves-firmware\lucidgloves-firmware.ino:3:0:
HardwareConfig.hpp:14:1: error: 'LED' does not name a type
 LED led(PIN_LED);
 ^
C:\Users\AikoKiss\Downloads\Compressed\FlexiaGloves-main_2\FlexiaGloves-main\firmware\lucidgloves-firmware\lucidgloves-firmware.ino: In function 'void setup()':
lucidgloves-firmware:77:3: error: 'led' was not declared in this scope
   led.setup();
   ^
C:\Users\AikoKiss\Downloads\Compressed\FlexiaGloves-main_2\FlexiaGloves-main\firmware\lucidgloves-firmware\lucidgloves-firmware.ino: In function 'void loop()':
lucidgloves-firmware:89:5: error: 'led' was not declared in this scope
     led.setState(LED::State::BLINK_STEADY);
     ^
lucidgloves-firmware:92:5: error: 'led' was not declared in this scope
     led.setState(LED::State::ON);
     ^
exit status 1
'LED' does not name a type

Stephen-3DVR, 2022-04-13:
This looks like a compiler issue with the Flexia firmware, yes? It would be more appropriate to add an issue to **that** fork's issue tracker.

danwillm, 2022-04-14:
indeed, this looks to be an issue related to Exia's firmware. Hopefully you've got this sorted by now, but this isn't an issue to be had on the lucidgloves' repository.

Feel free to contact us on the discord server, or open an issue on Exia's firmware repository.

closed 2022-04-14 by danwillm