LucidVR/lucidgloves #85: Is this the right place to ask questions about parts?

Snailedlt, 2022-08-27:
I was looking for discussions, but notice you don't have any, so as a 2nd choice I opted for writing an issue instead, sorry if this is the wrong place to ask.

Anyways, I am looking through the parts list, and I see a few of the items don't have links, and some have amazon links only. I'm a software developer and have no experience with engineering or engineering parts, so please excuse my lack of knowledge in these things :)

Here's a list of the things I'm wondering about specifically:
1. The velcro straps. I don't know what to get, since I don't know the length or type to get, a link would be very helpful as a guideline.
2. The crimp set with only an amazon link, which has super expensive shipping and toll here in Norway. I was wondering if this is a good alternative to that: https://a.aliexpress.com/_mtOFRWo
3. The powerbank. I see the one you recommend is 20000 mAh, but do you really need that much? How long does 20000mAh last usually? Would prefer to have a smaller powerbank if it's viable.
4. MG90S doesn't have an aliexpress link, but I'm assuming this is the right one: https://a.aliexpress.com/_mL9ymPO
5. Found these retractable badge reels. Do they look good enough? https://a.aliexpress.com/_mrq9OXe

RyanTripleSeven, 2022-10-25:
Hey!

1. Velcro Straps.[These](https://www.aliexpress.us/item/3256801412856833.html?spm=a2g0o.productlist.0.0.6a2354f3xKrEzB&algo_pvid=426bb75c-4a9e-4993-b0fc-ebb2535b6b89&algo_exp_id=426bb75c-4a9e-4993-b0fc-ebb2535b6b89-4&pdp_ext_f=%7B%22sku_id%22%3A%2212000028724432205%22%7D&pdp_npi=2%40dis%21USD%213.98%210.01%21%21%21%21%21%40210318cb16667010499587731e59e6%2112000028724432205%21sea&curPageLogUid=JjcaHCtTgF7Q) should work, and you can select the size you need. Just measure the circumference of your wrist.
2. That crimping set looks fine
3. I guess you technically don't need 20000mAh, I'm using a 5000mAh after all, but it's good to have it last long.
4. those servos look fine.
5. So do the badge reels.

Snailedlt, 2022-10-25:
Thanks, this helps a lot! :)
How long is the battery life with your 5000mAh battery if you don't mind me asking?

RyanTripleSeven, 2022-10-25:
I haven't completely drained the batteries, as I always charge after use, but I've easily gotten 2 hours from it.

Snailedlt, 2022-10-26:
Alright, thanks! Then there's no need for anything larger than 10000mAh imo :)