LucidVR/lucidgloves #95: commulation problesms

dwj66, 2023-03-17:
​I have made a completely same glove (left) as your Prototype 4, but when I connected it with SteamVR (and vive focus 3) through serial communication, it always failed, and I cannot calibrate servo using the opengloves tools, can you help me with that?

danwillm, 2023-03-17:
Please head over to our discord server and open up a support ticket there along with some additional info, including log files from here: C:\Program Files (x86)\Steam\logs\vrserver.txt so we can better assist you.

dwj66, 2023-03-20:
Fri Mar 17 2023 16:44:09.586 - //==============================================================================================
Fri Mar 17 2023 16:44:09.586 - ================================================================================================
Fri Mar 17 2023 16:44:09.586 - ================================================================================================
Fri Mar 17 2023 16:44:09.586 - vrserver.exe 1.25.5 startup with PID=25504, config=D:\VR\steam\config, runtime=D:\VR\steaminstallation\steamapps\common\SteamVR
Fri Mar 17 2023 16:44:09.586 - [vrserver] watchdogs enabled
Fri Mar 17 2023 16:44:09.588 - CSharedResourceNamespaceServer starting up
Fri Mar 17 2023 16:44:09.588 - WaitNamedPipe SteamVR_Namespace failed because no one is listening at that name.
Fri Mar 17 2023 16:44:09.589 - CSharedResourceNamespaceClient::Init(): failed connect
Fri Mar 17 2023 16:44:09.589 - CSharedResourceNamespaceClient::Init(): still waiting for vrserver to reply with namespace data
Fri Mar 17 2023 16:44:09.589 - Set thread FFFFFFFFFFFFFFFE priority to: 1
Fri Mar 17 2023 16:44:09.589 - Set thread FFFFFFFFFFFFFFFE priority to: -15
Fri Mar 17 2023 16:44:09.633 - Connecting client (SteamVR_Namespace) app container status: 1
Fri Mar 17 2023 16:44:09.633 - CSharedResourceNamespaceServer sent namespace data: 25504
Fri Mar 17 2023 16:44:09.695 - Client (SteamVR_Namespace) app container state: 1
Fri Mar 17 2023 16:44:09.711 - Connecting client (SteamVR_Namespace) app container status: 1
Fri Mar 17 2023 16:44:09.711 - CSharedResourceNamespaceServer sent namespace data: 25504
Fri Mar 17 2023 16:44:09.711 - CSharedResourceNamespaceClient::Init(): received namespace data 25504
Fri Mar 17 2023 16:44:09.711 - Self queried namespace: 25504
Fri Mar 17 2023 16:44:09.711 - VR server 1.25.5 (v1678993434) starting up with config=D:\VR\steam\config
Fri Mar 17 2023 16:44:09.711 - VRServer will wait for VRMonitor before exiting
Fri Mar 17 2023 16:44:09.715 - Skipping external driver D:\VR\steam\steamapps\common\OpenGlove because it is not a directory
Fri Mar 17 2023 16:44:09.716 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.716 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.716 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.717 - [Settings] Load Default Json Settings from D:\VR\VIVE Business Streaming\RRDriver\htc_business_streaming\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.717 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.717 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.718 - [Settings] Load Json Settings from D:\VR\steam\config\steamvr.vrsettings
Fri Mar 17 2023 16:44:09.722 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.722 - [Settings] Load Default Json Settings from D:\VR\VIVE Business Streaming\RRDriver\htc_business_streaming\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.722 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.722 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.723 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.723 - [Settings] Load Default Json Settings from D:\VR\steaminstallation\steamapps\common\SteamVR\resources\settings\default.vrsettings
Fri Mar 17 2023 16:44:09.723 - [Settings] Load Json Settings from D:\VR\steam\config\steamvr.vrsettings
Fri Mar 17 2023 16:44:09.745 - Adding controller type vive
Fri Mar 17 2023 16:44:09.746 - Adding controller type rift
Fri Mar 17 2023 16:44:09.746 - Adding controller type indexhmd
Fri Mar 17 2023 16:44:09.746 - Adding controller type vive_controller
Fri Mar 17 2023 16:44:09.747 - Adding controller type oculus_touch
Fri Mar 17 2023 16:44:09.747 - Adding controller type knuckles
Fri Mar 17 2023 16:44:09.749 - Add Json firmware manifest from {htc}/firmware/manifest.vrfirmware
Fri Mar 17 2023 16:44:09.749 - Add Json firmware manifest from {indexcontroller}/firmware/manifest.vrfirmware
Fri Mar 17 2023 16:44:09.749 - Add Json firmware manifest from {indexhmd}/firmware/manifest.vrfirmware
Fri Mar 17 2023 16:44:09.749 - Add Json firmware manifest from {lighthouse}/firmware/manifest.vrfirmware
Fri Mar 17 2023 16:44:09.750 - Add Json firmware manifest from {tundra_labs}/firmware/manifest.vrfirmware
Fri Mar 17 2023 16:44:09.750 - Web language is schinese
Fri Mar 17 2023 16:44:09.750 - [Web] Starting web server on port 27062
[... cut ...]
Fri Mar 17 2023 16:44:11.045 - Controller (/user/hand/right/input/system) not available for input to chord, ignoring chord
Fri Mar 17 2023 16:44:11.045 - Determined this is a steamvr_input app.
Fri Mar 17 2023 16:44:11.046 - Couldn't find action /actions/system/in/ShowPowerMenu. Ignoring it.
Fri Mar 17 2023 16:44:11.046 - Determined this is a steamvr_input app.
Fri Mar 17 2023 16:44:11.093 - openglove: Attempting connection to device (\\.\COM7)
Fri Mar 17 2023 16:44:11.093 - openglove: Received error connecting to port (\\.\COM7) - Error: 系统找不到指定的文件。

dwj66, 2023-03-20:
> Please head over to our discord server and open up a support ticket there along with some additional info, including log files from here: C:\Program Files (x86)\Steam\logs\vrserver.txt so we can better assist you.

> Please head over to our discord server and open up a support ticket there along with some additional info, including log files from here: C:\Program Files (x86)\Steam\logs\vrserver.txt so we can better assist you.

the log shows there is an Received error connectiong to the port, while i am 100% sure that the COM7 is the right port, and the communication between gloves and arduino is totally OK through COM7

danwillm, 2023-03-20:
The formatting of the string is invalid. It should be `\\.\COM7`

dwj66, 2023-03-20:
> 

appreciate for your response, I have actually tried many different formate of the string, including "\\.\COM7", but none of them works. the Log txt shows:
(
Fri Mar 17 2023 16:44:12.093 - openglove: Attempting connection to device (\\.\COM7)
Fri Mar 17 2023 16:44:12.093 - openglove: Received error connecting to port (\\.\COM7) - Error: 系统找不到指定的文件。

Fri Mar 17 2023 16:44:12.805 - [OnMsg_LaunchApplication( openvr.tool.steamvr_environments )] from vrstartup (6872)
Fri Mar 17 2023 16:44:13.093 - openglove: Attempting connection to device (\\.\COM7)
Fri Mar 17 2023 16:44:13.093 - openglove: Received error connecting to port (\\.\COM7) - Error: 系统找不到指定的文件。

Fri Mar 17 2023 16:44:13.505 - SetApplicationPid appkey=openvr.tool.steamvr_environments pid=27404, external transition
Fri Mar 17 2023 16:44:13.505 - SetApplicationPid: Setting app openvr.tool.steamvr_environments PID to 27404
Fri Mar 17 2023 16:44:13.505 - Processing message VRMsg_LaunchApplication from vrstartup (6872) took 0.7 seconds
Fri Mar 17 2023 16:44:14.094 - openglove: Attempting connection to device (\\.\COM7)
Fri Mar 17 2023 16:44:14.094 - openglove: Received error connecting to port (\\.\COM7) - Error: 系统找不到指定的文件。

Fri Mar 17 2023 16:44:14.278 - Connecting client (SteamVR_Namespace) app container status: 1
Fri Mar 17 2023 16:44:14.278 - CSharedResourceNamespaceServer sent namespace data: 25504
Fri Mar 17 2023 16:44:14.284 - Connecting client (VR_ServerPipe_25504) app container status: 1
)
the good news is that I changed another computer today, which is OK to connecte the gloves through a serial communation, but there is no qualified GPU to run the VR game, I will try to use other suitable computer tomorrow.

dwj66, 2023-03-21:
Today I used a new computer to run opengloves, the the serial communication and GPU are both OK, but still, it failed finally. What a build is left glove, and in the Steam VR hall, I can see both left hand and right hand, but only right hand is controllable using right handle, as for the left hand, it can moved by the left handle, but cannot be controlled by the left glove or left handle. and when I get into the opengloves interactive game, only right glove exist, and there is no left glove at all.
the vrserver.txt is:
`
Tue Mar 21 2023 14:21:41.173 - //==============================================================================================
Tue Mar 21 2023 14:21:41.173 - ================================================================================================
Tue Mar 21 2023 14:21:41.173 - ================================================================================================
Tue Mar 21 2023 14:21:41.173 - vrserver.exe 1.25.6 startup with PID=14872, config=D:\Program Files (x86)\Steam\config, runtime=D:\Program Files (x86)\Steam\steamapps\common\SteamVR
Tue Mar 21 2023 14:21:41.173 - [vrserver] watchdogs enabled
Tue Mar 21 2023 14:21:41.175 - CSharedResourceNamespaceServer starting up
Tue Mar 21 2023 14:21:41.175 - Set thread FFFFFFFFFFFFFFFE priority to: 1
Tue Mar 21 2023 14:21:41.175 - WaitNamedPipe SteamVR_Namespace failed because no one is listening at that name.
Tue Mar 21 2023 14:21:41.175 - CSharedResourceNamespaceClient::Init(): failed connect
Tue Mar 21 2023 14:21:41.175 - CSharedResourceNamespaceClient::Init(): still waiting for vrserver to reply with namespace data
Tue Mar 21 2023 14:21:41.175 - Set thread FFFFFFFFFFFFFFFE priority to: -15
Tue Mar 21 2023 14:21:41.275 - Connecting client (SteamVR_Namespace) app container status: 1
Tue Mar 21 2023 14:21:41.275 - CSharedResourceNamespaceServer sent namespace data: 14872
Tue Mar 21 2023 14:21:41.291 - Client (SteamVR_Namespace) app container state: 1
Tue Mar 21 2023 14:21:41.307 - Connecting client (SteamVR_Namespace) app container status: 1
Tue Mar 21 2023 14:21:41.307 - CSharedResourceNamespaceServer sent namespace data: 14872
Tue Mar 21 2023 14:21:41.307 - CSharedResourceNamespaceClient::Init(): received namespace data 14872
Tue Mar 21 2023 14:21:41.307 - Self queried namespace: 14872
Tue Mar 21 2023 14:21:41.308 - VR server 1.25.6 (v1679090868) starting up with config=D:\Program Files (x86)\Steam\config
Tue Mar 21 2023 14:21:41.308 - VRServer will wait for VRMonitor before exiting
Tue Mar 21 2023 14:21:41.314 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.314 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.314 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.315 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.315 - [Settings] Load Default Json Settings from C:\Users\Administrator\AppData\Local\HTC\Viveport\SteamVR\htc_rr\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.315 - [Settings] Load Default Json Settings from D:\Program Files (x86)\VIVE BUSINESS STREAM\VIVE Business Streaming\RRDriver\htc_business_streaming\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.315 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.316 - [Settings] Load Json Settings from D:\Program Files (x86)\Steam\config\steamvr.vrsettings
Tue Mar 21 2023 14:21:41.321 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.321 - [Settings] Load Default Json Settings from D:\Program Files (x86)\VIVE BUSINESS STREAM\VIVE Business Streaming\RRDriver\htc_business_streaming\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Default Json Settings from C:\Users\Administrator\AppData\Local\HTC\Viveport\SteamVR\htc_rr\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Default Json Settings from D:\Program Files (x86)\Steam\steamapps\common\SteamVR\resources\settings\default.vrsettings
Tue Mar 21 2023 14:21:41.322 - [Settings] Load Json Settings from D:\Program Files (x86)\Steam\config\steamvr.vrsettings
Tue Mar 21 2023 14:21:41.331 - Adding controller type vive
Tue Mar 21 2023 14:21:41.331 - Adding controller type rift
Tue Mar 21 2023 14:21:41.332 - Adding controller type indexhmd
[... cut ...]
Tue Mar 21 2023 14:21:49.794 - htc_business_streaming: [1, 5712] The texture set handle is [0] 2147499650, [1] 2147499714, [2] 2147499778
Tue Mar 21 2023 14:21:49.811 - htc_business_streaming: Update new sync texture 2147491970
Tue Mar 21 2023 14:21:49.889 - htc_business_streaming: VREvent_SceneApplicationChanged
Tue Mar 21 2023 14:21:49.891 - htc_business_streaming: The app[5712][0][5712, 0] is begin!
Tue Mar 21 2023 14:21:50.001 - htc_business_streaming: Update new sync texture 1073754178
Tue Mar 21 2023 14:21:50.449 - htc_business_streaming: Update new sync texture 2147491970

`

dwj66, 2023-03-21:
> Please head over to our discord server and open up a support ticket there along with some additional info, including log files from here: C:\Program Files (x86)\Steam\logs\vrserver.txt so we can better assist you.
Today I used a new computer to run opengloves, the the serial communication and GPU are both OK, but still, it failed finally. What a build is left glove, and in the Steam VR hall, I can see both left hand and right hand, but only right hand is controllable using right handle, as for the left hand, it can moved by the left handle, but cannot be controlled by the left glove or left handle. and when I get into the opengloves interactive game, only right glove exist, and there is no left glove at all.

[vrserver2.txt](https://github.com/LucidVR/lucidgloves/files/11025880/vrserver2.txt)