LucidVR/lucidgloves #101: Can't get lucidgloves to work with steamVR

m4tustechna, 2023-05-08:
Hello. I recently put together LucidVR LucidGloves Prototype 4. I assembled the virtual reality gloves according to "How to build cheap VR Haptic Gloves to FEEL VR." video by Lucas. I'm now on the part where I have to set up the driver to work with SteamVR.

In the Arduino IDE, when I open the Serial Monitor, I can see the data correctly and I know the potentiometers on the gloves work. I've installed OpenGloves from steam and set the data correctly (I hope). When I start SteamVR, I can see the glove icons, which are greyed out. The opengloves driver doesn't emulate the Steam Index drivers. 

I also want to mention that instead of Oculus controllers I have HTC Vive Trackers and instead of arduino nano I have ESP-WROOM-32. ESPs are set to bluetooth.

One thing I know is that the ESP32 communicates with the OpenGloves controller. When I click "functions" > "servo testing...", I can move the motors. They take about 30 seconds before I can press the button again, but they work. 

I don't know what to do and I really need help, if possible please answer me asap

m4tustechna, 2023-05-08:
[vrserver.txt](https://github.com/LucidVR/lucidgloves/files/11422661/vrserver.txt)

m4tustechna, 2023-05-08:
I managed to figure out the next step I was supposed to do. I had to change the override data. 

Followed this : https://github.com/LucidVR/opengloves-driver/wiki/Configuring-the-Driver#tracking-overrides

Changed in Opengloves UI and manually, located at /steamapps/common/OpenGlove/resources/settings/default.vrsettings

Now, instead of grey, my gloves are flashing and SteamVR says "searching". Web console says it's paired with a bluetooth device

`Mon May 08 2023 18:50:51.518 - openglove: Bluetooth Device found (lucidgloves-right)
Mon May 08 2023 18:50:51.518 - openglove: Bluetooth Device is authenticated (lucidgloves-right)
Mon May 08 2023 18:50:52.569 - openglove: Connected to bluetooth (lucidgloves-right)`

It's progress but I still can't pair the gloves

lucas-vrtech, 2023-05-09:
It sounds like you don't have your vive trackers configured properly. Instead of using the override feature, try configuring your vive trackers as left and right hands. Under Manage Vive Trackers, set them as left and right hands.

m4tustechna, 2023-05-09:
I got it! I found out what the problem was. The problem was that I hadn't set the trackers to behave like controllers. I did try this before I wrote this issue on github, but I failed. The reason I failed is because I was trying to configure the reciever and not the tracker. Thanks Lucas for the pointers, that there was no fault other than that I set the trackers incorrectly

I might as well share what I have done for those who are still lost.

I used this post on reddit : https://www.reddit.com/r/beatsaber/comments/bn6e4u/ragesaqs_school_of_making_a_vive_tracker_maul/
Follow only part 1!

Unplug any VR devices and plug trackers directly with USB cable (make sure you can use those cables for data transfer and that they are not just for charging)
Just in case, make backup of the .json file. 
open the opengloves driver, reset the settings and then enter the serial number of the tracker in "Device Driver Emulation"

closed 2023-05-09 by m4tustechna