LucidVR/lucidgloves #102: Linux support?

ARPavel22, 2023-05-26:
Hi, thank you for this big work!

Now im trying to compile Steam VR drive for LINUX Platform.
I have some succes.
But need some work to move Windows functions like NAMED PIPE and othrer.

Maybe you or anyone Developer made it? 
[image]

ARPavel22, 2023-05-31:
I managed! In general, I will answer my own question. Yes, open glove works on linkuks. Ubuntu 22.04. It only took to cut out all the extra stuff like bluetooth and serial and rewrite the pipe system.

ARPavel22, 2023-05-31:
[image]

ARPavel22, 2023-05-31:
[image]

danwillm, 2023-05-31:
Nice, feel free to open a PR to the opengloves repo with your changes!