LucidVR/lucidgloves #107: gloves dont connect

ny1u, 2023-06-26:
I use esp32 , with a ch340c converter. The firmware was successful, changed to bluetooth connection, but when I try to connect the gloves writes that there is no response from the device. Tried to connect by wire , also nothing works. I think the problem is in the ch340c.
[image]

LittleNyanCat, 2023-06-26:
If the firmware was successfully flashed, then that means the CH340C worked without any issue. Bluetooth is completely independent of the hardware serial interface so it has nothing to do with the CH340C either...
Is Bluetooth dongle is version 4.0? Newer versions are known to cause issues with the ESP32

ny1u, 2023-06-26:
Dongle version is 5.0

LittleNyanCat, 2023-06-26:
That is a probably cause for the Bluetooth issue, however you should still be able to test it in serial mode (via USB cable)