LucidVR/lucidgloves #115: Quest 3?

TheBlueOompaLoompa, 2023-10-27:
I'm part of a group that's looking to build a set of these gloves for a project. We don't know if these work on quest 3. I'm guessing the shape is close enough where the mount will still work, but I'm worried about tracking. From what I read, it looks like they use combination hand tracking and a few IR light tracking points. I'm concerned it won't be able to track when the lights are occluded because hand tracking may not work. Has anyone tested this yet?

MasterSiu, 2023-10-27:
It doesn't work outside steam vr at the moment

TheBlueOompaLoompa, 2023-10-27:
I should clarify, I am writing a custom driver to connect it to quest headsets in my unity project, but I was wondering if someone had tested it in Steam VR with quest link. Specifically because the new method for controller tracking seems to rely on hand tracking to fill in for when the tracking lights were occluded.

danwillm, 2023-10-27:
When using the steamvr driver we've developed for the gloves, quest 3 is supported fine.

closed 2023-10-27 by danwillm