LucidVR/lucidgloves #116: Serial mointer working on v4 but not v5

yungfresh1hub, 2023-10-29:
Hi, I am making the v5, a few months back when I began, I flashed the v5 firmware to the esp32 then it worked. But now the serial monitor updates but the data does not chage, I have tried reflashing other esp32 but nothing worked besides the buttons. But when I flash the v4 firmware the serial motor updates like it should. Any fix? also my hall sensor and mux are working as the should when i test them in other projects

m4tustechna, 2023-12-23:
Excuse me, but where did you download the v5 firmware from ?
I am also working on LucidGloves prototype 5 but I can't find where I can download the new firmware. Is it on discord ?

giovannylimin, 2024-02-20:
Hi, do you remember which v5 firmware that worked? Thank you