LucidVR/lucidgloves #120: Adding IR LED's around wrist to provide quest tracking

JoelsephBW, 2024-01-04:
Any chance this could be implemented? See title. I'm thinking quest 2/3 if it's possible had it based tracking for the controllers, just making it track the wrist should provide a significant uptick in positional tracking right? Or maybe wrist+ back of the hand and maybe a different pattern for the palm? Maybe even finger tips? Or is that just pointlessly redundant? Since it already estimated finger position off of the potentiometer? I mean it might provide better splay sensing too... 

Tldr:
 would ir leds be an option to implement tracking in quest 2/3?

And how would I go about getting native non PC implementation of this if it will hypothetically work?

valzzu, 2024-01-13:
Dont think so

RedHawk989, 2024-03-25:
It's not possible to do that. (Meta is too locked down for this to happen natively). most you could do is take apart a controller, but you still would need to maintain it's IR LED placement geometry. 

You _could_ attempt a pcvr custom solution to do this, but you'd need separate cams wired to a PC, and lots of time to tune algorithms, make controller drivers, and figure stuff out. Doesn't make sense for this project to attempt on its own.

AthemiS13, 2024-12-12:
you can use hand tracking