LucidVR/lucidgloves #129: I designed an updated model.

Yang66g, 2025-07-19:
I am a 17-year-old student in China. I have designed a new prototype based on an earlier one, which connects with a buckle to the hand, making it more convenient to wear and also looks more elegant, with great potential for development. However, the problem is that due to a busy academic schedule, I do not have the time to improve it and can only keep it as a design on paper (I am not good at modeling, programming, etc.; design is my strength). Is there anyone interested in helping me?

JamieStar-was-taken, 2025-07-20:
Could you please provide the design on paper

Yang66g, 2025-07-20:
> 您能否提供纸上的设计
当然可以，但请给我一点时间

Anxin10, 2025-08-07:
I'm also from China. Would it be possible to connect more easily for communication?

Yang66g, 2025-08-07:
哥们你直接说中文呗，微信GTX690xx

Yang.
***@***.***

学生
中国

AlankejiBPB, 2026-09-04:
或许咱们可以交流一下，我最近也在做这个项目